openhwfoundation/cv32e40s #12: Study (and if needed re-introduce) separate mret_insn_o/mret_dec_o and uret_insn_o/uret_dec_o signals

Silabs-ArjanB, 2021-04-14:
The original CV32E40P decoder differentiates between insn_o and dec_o signals for mret, uret and dret. This distinction was removed on the CV32E40X.

On the CV32E40P the following modification proves SEC clean when PULP_SECURE = 0, but not SEC clean when setting PULP_SECURE=1.

```
               begin
                 illegal_insn_o = (PULP_SECURE) ? current_priv_lvl_i != PRIV_LVL_M : 1'b0;
                 mret_insn_o    = ~illegal_insn_o;
-                mret_dec_o     = 1'b1;
+                mret_dec_o     = ~illegal_insn_o;
               end
 
               12'h002:  // uret
               begin
                 illegal_insn_o = (PULP_SECURE) ? 1'b0 : 1'b1;
                 uret_insn_o    = ~illegal_insn_o;
-                uret_dec_o     = 1'b1;
+                uret_dec_o     = ~illegal_insn_o;
               end
 
               12'h7b2:  // dret
               begin
                 illegal_insn_o = !debug_mode_i;
                 dret_insn_o    =  debug_mode_i;
-                dret_dec_o     =  1'b1;
+                dret_dec_o     =  debug_mode_i;
               end
```

Only making the dret_dec_o related change is SEC clean for both PULP_SECURE = 0 and PULP_SECURE = 1.

For the CV32E40S the above non-equivalency needs to be understood and maybe the original distinction between insn_o and dec_o signals for mret and uret needs to be re-introduced.

Silabs-ArjanB, 2021-05-05:
uret will not be need at all

silabs-oysteink, 2021-11-05:
The _dec signales were introduced in commit 2108fd4a8cf06b2bc51a761ad0e362c426a431cd, with a message containing ":bug: fix illegal vs xret race condition"

In E40P, the privilege level will change **before** the associated jump is taken (privilege level is updated in FLUSH_WB, and jump is taken in state XRET_JUMP, which follows FLUSH_WB). If the privilege level changes from M to U due to an mret, the mret_insn_o will be deasserted once the FSM reaches XRET_JUMP and no jump will be performed (unique case may also be violated). Letting mret_dec_o be 1 regardless of privilege level allows the mret to successfully perform it's jump after having changed the privilege level.

In the CV32E40S, the jump due to an mret will be taken from ID, and the privilege level will not be changed until the mret reaches WB. The IF stage gets the new privilege level forwarded when mret is in ID, so that it will use the correct privilege level while fetching instructions at the mret jump target. 

Conclusion is that the mret_dec_o signal is not needed for CV32E40S as it handles mret instructions differently within the pipeline and controller.
[image]

closed 2021-11-08 by Silabs-ArjanB