openhwfoundation/cv32e40s #33: processor does not clear mpp when executing mret with user mode present

strichmo, 2021-07-25:
## Component
**Component:RTL**

### Description
The E40S leaves the MPP field in mstatus as M-mode (3) instead of U-mode (0) when executing a *ret instruction.  Since the E40S supports user mode, this functionality should be changed from the older M-mode-only E40S and E40P.

From the Privileged Spec 3.1.6.1
```
The MRET, SRET, or URET instructions are used to return from traps in M-mode, S-mode, or
U-mode respectively. When executing an xRET instruction, supposing xPP holds the value y, x IE
is set to x PIE; the privilege mode is changed to y; x PIE is set to 1; and xPP is set to U (or M if
user-mode is not supported).
```

### Steps to Reproduce
The original failure used Xcelium to detect.  This should fail with any simulator however.
Use the master branch of core-v-verif
e31f6c33e6f5bcba9c1aac69e2c2c465598729b9
% makeuvmt test TEST=riscv_arithmetic_basic_test_0 CV_CORE=cv32e40s SEED=1 

The test should fail with this message:
UVM_ERROR @ 342.300 ns : uvme_cv32e40s_core_sb.sv(371) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 24, pc: 0x00009386, csr: mstatus, rvfi = 0x00001880, rvvi = 0x00000080
UVM_ERROR @ 360.300 ns : uvme_cv32e40s_core_sb.sv(371) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 25, pc: 0x000000c2, csr: mstatus, rvfi = 0x00001880, rvvi = 0x00000080
UVM_ERROR @ 372.300 ns : uvme_cv32e40s_core_sb.sv(371) uvm_test_top.env.core_sb [CORESB] CSR Mismatch, order: 26, pc: 0x000000c6, csr: mstatus, rvfi = 0x00001880, rvvi = 0x00000080

Silabs-ArjanB, 2021-07-26:
Hi @strichmo Thanks for filing this issue. As you know user mode has not been implemented yet, but we can prioritize the effect of *RET instructions in a next sprint.

strichmo, 2021-07-26:
Thanks @Silabs-ArjanB one thing to keep in mind: The setting of user mode is one of the few setting that CANNOT be disabled on OVPSIM.  Thus when Lee recently enabled that for the E40S we saw these failures.  I would hope that with just one or a few prioritized fixes we could continue making progress as tests should not be purposely enabling user mode under the RTL is ready for it.

strichmo, 2021-12-07:
This has long been fixed and should have been closed.  I am closing now.

closed 2021-12-07 by strichmo