openhwfoundation/cv32e40s #92: Dummy instruction inserted too early after CSR write

Silabs-ArjanB, 2022-01-04:
After a CSR write to cpuctrl, the dummy instruction insertion does not observe the written dummy instruction interval (as a dummy instruction can get inserted immediately violating the 1-N interval requirement).

The a_no_count_overflow assertion masks this issue because of it use of rnddummyfreq_written, see https://github.com/openhwgroup/cv32e40s/blob/6f42e427c296120d5a6c1367e8eab574d5c0bbc3/sva/cv32e40s_dummy_instr_sva.sv#L62

All rnddummyfreq_written logic should be removed from that assertion file.

silabs-oysteink, 2022-01-04:
[image]

In this case, the cnt_q has already passed the new value for lfsr_cnt after the CSR write, creating a dummy instruction immediately after the write. I guess the cnt_q should be reset for these writes as well.

Silabs-ArjanB, 2022-01-11:
Can the following also be done?

- Add an id_stage assertion that for dummy instructions operand_a_fw_mux_sel equals SEL_LFSR, or operand_a_fw_mux_sel equals SEL_REGFILE and the read address is 0. Same for operand_b_fw_mux_sel.
- In the register file make sure that REGFILE_NUM_WRITE_PORTS is taken into account also if SECURE=1 (currently it is not taken into account as can for example be seen by the use of we_dec[0][0])

closed 2022-01-21 by Silabs-ArjanB