openhwfoundation/cv32e40s #148: Illegal concatenation in bhv/rvfi_sim_trace.sv

MikeOpenHWGroup, 2022-02-23:
### Component
`bhv/rvfi_sim_trace`

### Description of Issue
At [line 108](https://github.com/openhwgroup/cv32e40s/blob/a6ead13ddacdca83516db81240ce11d94d1a9c0a/bhv/rvfi_sim_trace.sv#L108) there is an attempted concatenation of a bit-vector (`tmp_trace.asm`) and a string (`asms[i]`).

### Steps to Reproduce
In theory, any attempt to compile the RTL with the tracer included should do it.  I typically do this from `core-v-verif` under dsim:
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32e40s/sim/uvmt
$ make test TEST=hello-world SIMULATOR=dsim
```
Note: the above assumes you have the [required shell environment variables](https://github.com/openhwgroup/core-v-verif/tree/master/mk#required-corev-environment-variables) pointing to your local toolchain.

### Additional Context
The Tracer fails compilation under Metrics dsim.  I have not been able to try it with any other simulator.

MikeOpenHWGroup, 2022-02-24:
I have implemented a "fix" for this issue locally by changing the definition of `tmp_trace_t::asm` from a bit vector to a string on [line 43](https://github.com/openhwgroup/cv32e40s/blob/a6ead13ddacdca83516db81240ce11d94d1a9c0a/bhv/rvfi_sim_trace.sv#L43):
```
bit [32*16-1:0]  asm;
```
to:
```
string  asm;
```
I _think_ that should do it and with this change a local regression passes, but I am not familiar enough with the code to know for certain that this is a safe change.

silabs-halfdan, 2022-02-24:
The issue with that is that we are not able to probe strings.
I was able to recreate this with vsim, but xrun is happy with it and doesn't even give a warning.

silabs-halfdan, 2022-02-24:
I just made a PR that fixes the issue for vsim at least, and I would expect the same to work for dsim.
https://github.com/openhwgroup/cv32e40x/pull/457

MikeOpenHWGroup, 2022-02-24:
> I just made a PR that fixes the issue for vsim at least, and I would expect the same to work for dsim. [openhwgroup/cv32e40x#457](https://github.com/openhwgroup/cv32e40x/pull/457)

Thanks.   Works with dsim.

MikeOpenHWGroup, 2022-02-24:
[openhwgroup/cv32e40x#457](https://github.com/openhwgroup/cv32e40x/pull/457) also needs to be applied to the E40S.

Silabs-ArjanB, 2022-02-24:
Yes, this will be part of our regular 40X -> 40S merge eventually

silabs-robin, 2022-11-02:
Can this be closed?

silabs-oivind, 2022-11-02:
Yes. This has been merged to e40s a long time ago.

closed 2022-11-02 by Silabs-ArjanB