openhwfoundation/cv32e40s #196: Wrong D-mode priv lvl

silabs-robin, 2022-05-13:
### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
40s hash `fdbba032`.

Trace with CEX:
[image]
This trace is long, but the color-highlights (RGB tresis) show the problem.
The core executes in debug mode, while the privilege mode is U-mode, and this is not allowed.

Debug spec:
[image]

Reproduction:
```
cover -name mycov_dmodeumode {
  ((dut_wrap.cv32e40s_wrapper_i.rvfi_i.rvfi_valid == 1'b1)
  && (dut_wrap.cv32e40s_wrapper_i.rvfi_i.rvfi_dbg_mode == 1'b1)
  && (dut_wrap.cv32e40s_wrapper_i.rvfi_i.rvfi_mode == PRIV_LVL_U))
};
```
This cover should be an ok starting point.

silabs-robin, 2022-05-13:
@Silabs-ArjanB @silabs-oysteink

Silabs-ArjanB, 2022-07-18:
@silabs-oysteink Is all code execution in debug mode 'due to abstract commands'? Note that section A.2 of the debug spec states "This implementation only implements the Access Register abstract command for GPRs on a halted
hart, and relies on the Program Buffer for all other operations."

Silabs-ArjanB, 2022-07-25:
@silabs-robin I filed https://github.com/riscv/riscv-debug-spec/issues/744 to get a better understanding about what is really the requirement here.

Silabs-ArjanB, 2022-07-26:
Should be fixed with https://github.com/openhwgroup/cv32e40s/pull/268

silabs-robin, 2022-07-27:
The issue has been fixed.
I have an assertion ("a_dmode_mmode") which fails before applying the fixes, and passes after applying the fixes.

(All possible core configurations have _not_ been tested, but they will be when my assertions are merged into core-v-verif and regressions are ran.)

closed 2022-07-27 by silabs-robin