openhwfoundation/cv32e40s #206: Undefined behavior for mscratchcsw and mscratchcswl not handled in RTL

silabs-hfegran, 2022-05-30:
## Undefined behavior for mscratchcsw and mscratchcswl not handled in RTL
Behavior of mscratchcsw and mscratchcswl is undefined in the cases mentioned below (CLIC spec chapter 13.2 and 14.1). As with other undefined/reserved behavior, we should trigger illegal instruction exceptions.

### Background
CLIC spec states as follows:
>13.2 (...) In practice, only the read-modify-write (swap/CSRRW) operation is useful for xscratchcsw. In contrast, the "pure read" and "pure write" operations are meaningless and not useful at all. The implementation of the non-CSRRW variants (when either rd or rs1 is x0 or when rs1 is an immediate operand) on xscratchcsw is not defined/reserved.

> 14.1 (...) Similar to xscratchcsw, only the read-modify-write (swap/CSRRW) operation is defined for xscratchcswl. The implementation of the non-CSRRW variants (when either rd or rs1 is x0 or when rs1 is an immediate operand) on xscratchcswl is not defined/reserved.

Silabs-ArjanB, 2022-10-14:
@silabs-oysteink Has this been fixed now?

silabs-oysteink, 2022-10-14:
Yes, all CSR instructions except CSRRW with rd!=0 and rs1!=1 will be flagged as illegal.

Silabs-ArjanB, 2022-10-14:
@silabs-hfegran This should be fixed now (both in our user manual and RTL). Please close if agreed.

silabs-hfegran, 2022-10-14:
Looks good to me

closed 2022-10-14 by silabs-hfegran