openhwfoundation/cv32e40s #236: rvfi wrong rs1_rdata

silabs-robin, 2022-06-20:
### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)
**Component:Other**: For any other issues

### Steps to Reproduce
In formal, can just 1) look at the waveform and write covers/asserts based on that, or 2) can use my branch of core-v-verif with an assertion that catches this, 3) possibly the same issue exists in the failing rvfi fv jobs.
(Opt 2 is probably the safest bet.)

My branch `rvfi_assert` is shown at https://github.com/openhwgroup/core-v-verif/compare/cv32e40s/dev...silabs-robin:rvfi_assert

The problem is that `rvfi_rs1_rdata` does not show the "pre-state" (aka "before" data).
The trace below shows `rvfi_valid` at `rvfi_order = 1`, and with the `RF_REG_RV` the pre-state of the registers should be 0.
But `rs1_rdata` seems to be using data from `rd_wdata`, seemingly because of `last_op`.

[image]

[image]

silabs-oysteink, 2022-07-01:
PR #241 should hopefully fix this issue.

silabs-robin, 2022-07-01:
It seems the problem persists.
I am at `b8de6d` of your cv32e40s branch, and I still get a CEX on the `a_rs1_resetvalue` assert from my `rvfi_assert` core-v-verif branch.
The new text `Only update rs1/rs2 on the first part of a multi operation instruction` is visible in the elaborated rvfi_i module so it should be the latest RTL.

Could it be that I need to update my core-v-verif branch with some particular configuration?

It fails on an illegal instruction, giving rs1_rdata on the very first rvfi_valid:
[image]

silabs-oysteink, 2022-07-04:
There is a second issue that is causing this, after fixing the RVFI issue. A secure jump will perform two operations, and currently the controller is able to kill the jump (or branch) after only one of two operations has retired. This kill should not have happened, as the state will be updated without the instruction being fully executed. A fix for this will be introduced on the E40X when the new 'first_op' signal is fully integrated.

silabs-oysteink, 2022-07-07:
With PR #245 this should finally be fixed. Still waiting for formal to converge, I will add a comment once it is done.

silabs-robin, 2022-11-02:
I think we can close this.

`a_rs1_resetvalue` has a bound of 37 in my run right now.
So the initial problem is definitely solved.
If there are further problems then it should get caught by long-running regression jobs and we'll open a new ticket.

If there are any objections then just re-open this.

closed 2022-11-02 by silabs-robin

silabs-robin, 2022-11-02:
_(Both asserts converged now, btw)_