openhwfoundation/cv32e40s #242: Wrong debug mode handling for User mode

tzwaenn, 2022-07-04:
# Template for Bug Issue

### Component

**Component:RTL**

### Steps to Reproduce
See attached trace
[image]
We see the core being set up for user mode, going to debug mode, and then executing a dret instruction. The dret violates the specification since it is supposed to restore the privilege from dcsr.prv which is not happening since dcsr.prv is Machine mode, but privilege level stays User mode. The RTL seems to try to set the privilege mode by assigning priv_lvl_n, but not priv_lvl_we, so the restore is not happening.
Incidentally, the trace also shows a previous issue: Upon entering debug mode, dcsr.prv is supposed to be set to the current privilege level (here User mode), but it is tied to Machine mode in RTL.

The issue was found with the OneSpin Processor Verification App, specifically the assertion verifying return behavior.

Silabs-ArjanB, 2022-07-04:
additional ionfo from OneSpin: Saving the mode in dcsr.prv constantly stores PRIV_LVL_M in line 1498 and on a dret, prv_lvl_n is set as expected in line 1556, but not priv_lvl_we, so no level transition is happening.

Silabs-ArjanB, 2022-07-18:
Both reported issue should be fixed with https://github.com/openhwgroup/cv32e40s/pull/260.

@tzwaenn Thank you very much for finding and reporting these issues (the user mode related additions to the debug circuitry were still work in progress, but it is great to see that OneSpin catches these issues). Could you please run OneSpin again and check if these issues have now properly been fixed?

tzwaenn, 2022-07-20:
There is no longer a fail for the return assertion.

Silabs-ArjanB, 2022-07-20:
@tzwaenn Could you please close this issue if you agree it has been fixed?

closed 2022-07-20 by tzwaenn