openhwfoundation/cv32e40s #248: "rvfi_dbg" wrong cause

silabs-robin, 2022-07-08:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)
**Component:Other**: For any other issues

### Description

On the first instr in dmode, `rvfi_dbg` _"contains the debug cause"_.
Also, `dcsr.cause` _"Explains why Debug Mode was entered"_
AFAICT, they should be the same on the first instr in dmode.
Problem: They are not. (See screenshot below.)

[image]

The trace above shows the core entering dmode because of `debug_req_i`.
This cause is correctly reflected in `dcsr.cause` (3 is "haltreq").
But rvfi incorrectly shows the cause to be "1", because it has some special handling of "ebreak".

### Steps to Reproduce

1. Use my fork and branch https://github.com/silabs-robin/core-v-verif/tree/rvfi_assert, hash `f9b040d4`.
2. Run formal and check `a_dbg_cause`.

Silabs-ArjanB, 2022-07-18:
This is both an RVFI issue and an assertion issue.

The RVFI issue should be solved by https://github.com/openhwgroup/cv32e40s/pull/258.

The assertion issue relates to a misunderstanding of the rvfi_dbg signal. It does _not_ always have to be equal the debug cause in dcsr as shown in Table 27 Table of scenarios for first instruction of exception/interrupt/debug handler. Further clarification has been added in the user manual via https://github.com/openhwgroup/cv32e40x/pull/621.

Resolving this issue (as it tracks the design repos and not core-v-verif that still requires an assertion update).

silabs-robin, 2022-07-27:
> an ebreak in debug mode will be reported via rvfi_dbg

Why?
_"For the first instruction after entering debug, the rvfi_dbg signal contains the debug cause"_.
An ebreak in debug mode goes from D-mode to D-mode, so I don't see why that is "entering debug".

Silabs-ArjanB, 2022-07-28:
@silabs-robin This scenario is considered 'entering debug' because a branch is performed to dm_halt_addr_i.

silabs-robin, 2022-07-28:
The RVFI issue appears to have been fixed. CEXes now point to the need to update the assertion itself.

closed 2022-07-28 by silabs-robin