openhwfoundation/cv32e40s #249: Wrong execute permission with mseccfg.MML and no matching region

tzwaenn, 2022-07-11:
### Component

**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
When the first instruction after reset sets mseccfg.MML, the fetch of the next instruction actually goes through. However, the Smepmp spec states that for MML set, "Executing code from a region without a matching rule or with a matching S/U-mode-only rule is denied." With no PMPs set up, the whole memory should thus be denied execute access after setting MML. This is also nicely illustrated in the picture in section 2.2: setting MML is enough to deny execute rights to M mode without a matching region.
The issue was found with assertion RV_chk.ops.XCPT_IF_ID_a of the OneSpin Processor Verification App.

Silabs-ArjanB, 2022-07-18:
Hi @tzwaenn Thank you for finding and reporting this issue. https://github.com/openhwgroup/cv32e40s/pull/259 should have fixed this issue now. Could you please check if the issue has indeed been solved?

Silabs-ArjanB, 2022-07-18:
@tzwaenn Thank you for the info. Then I clearly did not fix the issue as I fixed an issue related to PMP_NUM_REGIONS=0. I will have another look (I am not too familiar with this code, so if it goes too deep I might have to postpone it).

@tzwaenn Will you be checking the bevaior for PMP_NIUM_REGIONS=0 as well (in that case the PMP is still present, but none of its regions can be programmed)

tzwaenn, 2022-07-18:
@Silabs-ArjanB: We so far mostly used NUM_PMP_REGIONS=0 for Zcb, Zcmb tests which all passed. But if mseccfg still exists if there are 0 regions, our Smepmp tests should cover that case as well (I guess having 0 regions will disable user mode as it is no longer possible to have memory regions accessed by user mode)

Silabs-ArjanB, 2022-07-18:
>> But if mseccfg still exists if there are 0 regions, our Smepmp tests should cover that case as well (I guess having 0 regions will disable user mode as it is no longer possible to have memory regions accessed by user mode)

0 regions should not disable user mode, but it indeed forces mseccfg hardwired to 0, which is probably why PMP_NUM_REGIONS did not trigger the same issue as you reported here.

@tzwaenn I pushed another fix (https://github.com/openhwgroup/cv32e40s/pull/261). Could you see again now? (I don't have a simulation setup for the given test case, so I am making the fix based on the ePMP spec only)

tzwaenn, 2022-07-19:
@Silabs-ArjanB what I meant about deactivating user mode in practice is that if 0 PMP regions takes the access fault computation from the n regions case, it will be impossible to execute any instruction in user mode as all user mode memory accesses are illegal and it is impossible to make them lega: no PMP entry can be added that could match the user mode address and a not matching access is always denied outside of M mode even without Smepmp.

tzwaenn, 2022-07-19:
The fix looks good - no more quick fails when firing up a handful of assertions.

Silabs-ArjanB, 2022-07-20:
@tzwaenn Could you please close this issue if you agree it has been fixed?

closed 2022-07-20 by tzwaenn