openhwfoundation/cv32e40s #257: UM: Wrong description of minimum PMP granularity

silabs-hfegran, 2022-07-15:
## Wrong description of minimum PMP granularity

Issue: minimum  granularity (G) tied to number of pmp regions implemented.

The user manual currently states that the minimum granularity, G, of a PMP regions is 2^(PMP_NUM_REGIONS + 2) bytes. 
If this were to be true, it would lead to a absurd cases such as a minimum granularity of 2^66 bytes in the case of 64 PMP regions.
I presume the correct should be 2^(PMP_GRANULARITY+2)

### Steps to Reproduce
Latest UM:
[image]

Silabs-ArjanB, 2022-07-15:
@silabs-hfegran Are you sure you are looking at the latest version as opposed to 0.2.0?

Silabs-ArjanB, 2022-07-15:
Had already been fixed

closed 2022-07-15 by Silabs-ArjanB