openhwfoundation/cv32e40s #270: xmsim: *E,ASRTST, uvmt_cv32e40s_pmp_assert.sv

silabs-kun, 2022-07-27:
# Error message

xmsim: *E,ASRTST (/work/kuzhu/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_pmp_assert.sv,728): (time 60379800 PS) Assertion uvmt_cv32e40s_tb.dut_wrap.cv32e40s_wrapper_i.core_i.load_store_unit_i.mpu_i.pmp_i.u_pmp_assert_lsu.gen_pmp_assert.a_accept_only_legal has failed 
stop: uvmt_cv32e40s_tb.dut_wrap.cv32e40s_wrapper_i.core_i.load_store_unit_i.mpu_i.pmp_i.u_pmp_assert_lsu.gen_pmp_assert.a_accept_only_legal = failed)

xmsim: *E,ASRTST (/work/kuzhu/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_pmp_assert.sv,782): (time 60379800 PS) Assertion uvmt_cv32e40s_tb.dut_wrap.cv32e40s_wrapper_i.core_i.load_store_unit_i.mpu_i.pmp_i.u_pmp_assert_lsu.gen_pmp_assert.a_nomatch_umode_fails has failed 
stop: uvmt_cv32e40s_tb.dut_wrap.cv32e40s_wrapper_i.core_i.load_store_unit_i.mpu_i.pmp_i.u_pmp_assert_lsu.gen_pmp_assert.a_nomatch_umode_fails = failed)

xmsim: *E,ASRTST (/work/kuzhu/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_pmp_assert.sv,788): (time 60379800 PS) Assertion uvmt_cv32e40s_tb.dut_wrap.cv32e40s_wrapper_i.core_i.load_store_unit_i.mpu_i.pmp_i.u_pmp_assert_lsu.gen_pmp_assert.a_uorl_onlyif_rwx has failed 
stop: uvmt_cv32e40s_tb.dut_wrap.cv32e40s_wrapper_i.core_i.load_store_unit_i.mpu_i.pmp_i.u_pmp_assert_lsu.gen_pmp_assert.a_uorl_onlyif_rwx = failed)

<br>**git commit**: https://github.com/silabs-kun/core-v-verif/commit/e7eb5f0335d2ffde9bbe42a49c5f04662defa046

### Steps to Reproduce
Please provide:
1. git hash: e7eb5f0335d2ffde9bbe42a49c5f04662defa046
2. make test TEST=pmp CFG=pmp GUI=1 WAVES=1 ADV_DEBUG=1
3. error message occur after 'switch mode', then one needs to click 'run' button on INDAGO to continue until the program finishes.

@silabs-robin

silabs-robin, 2022-07-27:
Thanks @silabs-kun. This seems very related to the other error, "a_deny_only_illegal".

silabs-robin, 2022-07-28:
I just noticed. This issue was opened on the wrong repo.
The errors above are from "core-v-verif", but this is the "cv32e40s" repo.