openhwfoundation/cv32e40s #272: PMP deny reaches bus?

silabs-robin, 2022-07-31:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)
**Component:Other**: For any other issues

### Steps to Reproduce

1. My branch [pmp_assert](https://github.com/silabs-robin/core-v-verif/tree/pmp_assert) at hash `139b7d4b`
2. Run formal, `tcf_cvverif`, with rc `pmp_g3r3`  (communicate in chat if necessary)

The waveform below shows:
1. Instruction fetch of `0x0040_1ffc` (`req` + `gnt`), with later rvfi retirement of that instruction
2. Immediately after the first fetch, the `+4` fetch of `0x0040_2000` starts (`req`)
3. When the first instruction executes, it updates `mseccfg.mmwp`, and this makes the second fetch "pmp-denied"

This problem is a bit tricky to judge, so I am not 100% sure, but please have a look and see if it is a real bug or not.
(Note also that the assert might need refinement, but I paused writing at the point of this CEX.)

[image]

Silabs-ArjanB, 2022-08-01:
Hi @silabs-robin Only judging on the above figure this does not seem like a violation (yet) to me. It sounds like that the assertion is forbidding certain **prefetches**, whereas the PMP should only prevent retirement of instructions using certain addresses. So PMP assertions should just look at retired instructions and since above wave shows only 1 rvfi_valid cycle there can never yet be a PMP related violation (assuming you did use the default reset values of the PMP CSRs).

silabs-robin, 2022-08-01:
Not a bug.

Explanation:
While data-side PMP error shall never reach the bus, instruction-side PMP errors can reach the bus for speculative pre-fetching _but then it must never be retired_.
The special instruction-side allowance is harmless when the PMA limits fetching to only Main Memory regions.

closed 2022-08-01 by silabs-robin