openhwfoundation/cv32e40s #275: Did we already fix https://github.com/lowRISC/ibex/issues/1719?

Silabs-ArjanB, 2022-08-03:
See https://github.com/lowRISC/ibex/issues/1719

silabs-robin, 2022-08-09:
I have made a cover property showing that we can exhibit correct behavior for the case in that issue.
(I left a todo to make an assert for this once we have reliable rvfi csr wdata signals.)

The original issue is confusing and I can't see clearly what they actually claim.
I tried to decipher their log, and it seems Spike is in the wrong and the Ibex rtl is right.
This is also confirmed by their comments further down, linking to a PR to Spike.

Ibex never had a problem, if we borrowed code properly then we never had a problem, and my cover indicates that we indeed do not have a problem?

[image]

```
setup-state:
5843330f

setup-equivalent:
1803130f

initial-cfg:
17978d0c

initial-locks:
00111100

initial-tor:
00001111

write-attempt:
5843330f

write-equivalent:
1803130f

write-locks:
00000000

result-cfg:
18978d0f

result-changed:
11000011  // Cfg0 changed as expected

spike-cfg:
18978d0c

spike-changed:
11000000  // Cfg0 refused to change
```

silabs-oivind, 2022-08-09:
Setting a region to TOR will only lock the address of the previous region (and not the config) in our implementation.

closed 2022-08-18 by Silabs-ArjanB