openhwfoundation/cv32e40s #281: Wrong "rvfi_mem_addr"

silabs-robin, 2022-08-19:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)
**Component:Other**: For any other issues

### Problem

RVFI reports two store instructions ("rvfi_mem_wmask"), while "rvfi_mem_addr" is the same on both.
However, the second addr seems to be a remnant from the previous store, not reflecting the actual address used on the second store.
This is shown in two ways, 1) calculating "rs1 + offset", and 2) looking at the write buffer inputs.
(See screenshot below.)

### Steps to Reproduce

* [My core-v-verif fork](https://github.com/silabs-robin/core-v-verif)
* Branch "pmp_assert_rvfi2"
* Hash "520ac237"
* In formal "tcf_cvverif"
* With rc "pmp_g3r3
* CEX on assert "a_noloadstore_musttrap"

[image]

Silabs-ArjanB, 2022-08-19:
At first glance RVFI does not take the write buffer into account. Once it does we will have a coverage hole in the sense that the logic in between the write buffer and the bus interface is no longer covered by RVFI. An issue or task should be added related to this.

Silabs-ArjanB, 2022-08-19:
Issue should be fixed on CV32E40X first and then merged over.

silabs-robin, 2022-10-04:
I plan on closing this one too once I have confirmed the fix.

silabs-robin, 2022-11-01:
Reran the aforementioned assert ("a_noloadstore_musttrap", which caught the original problem). Did not write a new assert specifically for this. The assert now has a bound of >=29 (CEX was at 21). This points to the problem being solved. Closing.

closed 2022-11-01 by silabs-robin