openhwfoundation/cv32e40s #283: pmpncfg.RWX not unchanged

silabs-robin, 2022-08-25:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce

I can provide a test case upon request, but my current setup has 2 pending rtl future-fixes in it and I think the case is visible just from source code vs spec.

### Problem

Spec:
_"The value of the collective R, W, X bitfield will remain unchanged when attempting to write R = 0 and W = 1 while mseccfg.MML = 0"_.

Source:
[image]

Example:
If you try to write the illegal value of `6`, the RTL will clear the two LSBs and leave the legal value of `4`, even if the original value was `0` and RWX was supposed to remain unchanged.
[image]

Silabs-ArjanB, 2022-08-25:
This issue is the same as https://github.com/openhwgroup/cv32e40s/issues/165.

Note that this requirement is not from the official RISC-V spec.

silabs-robin, 2022-08-27:
(I know this is a duplicate, but I want to leave it up until it is fixed, because then it is easier for me to track its status.)

silabs-oivind, 2022-10-03:
This is fixed in https://github.com/openhwgroup/cv32e40s/pull/290

Silabs-ArjanB, 2022-10-03:
@silabs-robin Please close this issue once you have confirmed that the fix is okay.

silabs-robin, 2022-11-01:
I believe it is fixed. Checked all asserts involving `rectify_cfg_write` in rc `pmp_pma_2` and `pmp_pma_4`, passing. Closing.

closed 2022-11-01 by silabs-robin