openhwfoundation/cv32e40s #286: Flush on MPRV-write?

silabs-robin, 2022-08-27:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)
**Component:Doc **: For issues in the Documentation (e.g. for User Manual or README.md files)

### Steps to Reproduce

1. 5f1cb75d, https://github.com/silabs-robin/core-v-verif/tree/pmp_assert
2. tcf_cvverif --rc=pmp_g3r3
3. a_noexec_cause

### Problem

Changing MPRV changes the "effective privilege mode", and hence also the decision of the PMP, but we don't flush the pipeline like we do for other PMP-related CSR updates, so it seems possible for instruction access faults to go undetected.

[image]

> PMP checks are applied to all accesses whose effective privilege mode is S or U, including instruction
fetches in S and U mode, data accesses in S and U mode when the MPRV bit in the mstatus register
is clear, and data accesses in any mode when the MPRV bit in mstatus is set and the MPP field
in mstatus contains S or U.
> -- privspec

> A CSR write to any of the PMP CSRs shall cause a flush of the CPU pipeline including its prefetch unit and potentially outstanding bus transactions.
> -- ipspec

Silabs-ArjanB, 2022-08-29:
According to the RISC-V priv spec:

> The MPRV (Modify PRiVilege) bit modifies the effective privilege mode, i.e., the privilege level
> at which loads and stores execute.
> 
> Instruction address-translation and protection are
> unaffected by the setting of MPRV.

It seems above description is looking at the instruction side PMP in relationship to MPRV. Are the assertions taking above quotes into account?

silabs-robin, 2022-08-29:
> Are the assertions taking above quotes into account?

No, that was an oversight. The model seems improperly connected here, I'll fix it.

It might have been that the text _"all accesses whose effective privilege mode is S or U, including instruction fetches"_ led me to believe that instructions fetches was included in the concept of "effective privilege mode", but looking at it again it is clear that MPRV only affects loadstores. (It is especially clear from _"Instruction address-translation and protection are unaffected by the setting of MPRV"_.)
Thanks for pointing it out.

closed 2022-08-29 by silabs-robin