openhwfoundation/cv32e40s #293: Dummy instruction frequency can be violated

silabs-krdosvik, 2022-08-30:
## Bug: The frequency of dummy instructions can be violated.

### Component: RTL dummy instructions

https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/xsecure.html
[image]
We want to ensure the shown requirements.

https://github.com/openhwgroup/cv32e40s/blob/master/rtl/cv32e40s_dummy_instr.sv
[image]

The code produces the following waveform:

[image]
In the clock tick that is set a ring around we observes that cnt_q > lfsr_cnt. Therefor we insert a dummy instruction by setting the signal dummy_insert_o high. The instruction to which a dummy has been inserted does not propagate further into the pipeline because (if_valid_o & id_ready_i) != 1. Nevertheless, we reset the counter.

An illustration of a possible counter example can be the following:
[image]
Where i = regular instruction, d = dummy instruction, b1, b2, b3 = branch instructions (which is also regular instructions) and dx = dummy instruction that will not enter the other pipeline stages. 
d is not actually reported in the rvfi stage, but if it was, it would look something like what’s shown above. The red symbols count the number of regular instructions in a row, 5, which is more than what we would expect, 4.

Conclusion:
The problem is that we reset the counter before making sure that the dummy instruction propagates into the pipeline. We encounter this when inserting the dummy instructions into invalid instructions, instead of valid instructions. 

Possible solutions:
1) Check that a dummy instruction was inserted into "valid" instructions. Can be done by changing the dummy_insert_o expression to dummy_insert_o = (cnt_q > lfsr_cnt) && dummy_en && instr_issues_i;
2) Change when you reset the counter. Can be done by changing the cnt_rst expression to cnt_rst = !dummy_en || (dummy_insert_o & instr_issued_i) || xsecure_ctrl_i.cntrst;
3) Other smart fixes.

Silabs-ArjanB, 2022-10-05:
Hi @silabs-krdosvik This issue should have been fixed now. Can you close the issue once you have confirmed that the fix is correct?

silabs-krdosvik, 2022-10-14:
[dummy_freq_resetting.pdf](https://github.com/openhwgroup/cv32e40s/files/9784328/dummy_freq_resetting.pdf)

Locks correct. Here is also my check for the scenario that we though would make the frequency requirement fail. It is in norwegian, so in case that is a problem just use google translate or so.

closed 2022-10-14 by silabs-krdosvik