openhwfoundation/cv32e40s #294: pmpncfg "keep old mode"

silabs-robin, 2022-09-01:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

## Problem

When G>=1 and one attemps to change to the unselectable mode NA4, according to our WARL rules we should keep the field's previous value, but we don't.

[image]

[image]

@silabs-oivind

Silabs-ArjanB, 2022-10-04:
@silabs-robin Can you please close this issue once you have confirmed the fix?

silabs-robin, 2022-11-01:
I believe it is fixed. Checked all asserts involving `rectify_cfg_write` in rc `pmp_pma_2` and `pmp_pma_4`, passing. Closing.