openhwfoundation/cv32e40s #300: rvfi_intr can be missed with clic shv-interrupt

silabs-hfegran, 2022-09-09:
## Missing rvfi_intr for clic selective hardware vectored interrupt
Under certain circumstances, a taken shv-interrupt does not result in correct rvfi_intr signalling. This appears to coincide with a write to mtvt as the last instruction retired before irq acknowledgement 

### Steps to Reproduce
Please provide:
1. git hash: 8ee7abf25c46
2. Formally analyzed assertion:
```
    property p_every_ack_followed_by_rvfi_intr;
      irq_ack ##1 rvfi_valid[->1]
        |->
         rvfi_intr.intr
      or
         rvfi_trap.debug
      or
         rvfi_trap.exception;
    endproperty : p_every_ack_followed_by_rvfi_intr
    
    a_every_ack_followed_by_rvfi_intr: assert property (p_every_ack_followed_by_rvfi_intr)
    else
      `uvm_error(info_tag,
        $sformatf("Every irq_ack should be followed by the corresponding rvfi_intr"));

```
4. Waves:
[image]

silabs-oysteink, 2022-10-05:
This issue should now be fixed with PR #316.

Silabs-ArjanB, 2022-10-05:
@silabs-hfegran Can you please close this issue if you agree it has been fixed. Also please note that the provided assertion is too relaxed: The following part should not be needed according to the specification of the involved signals:

```
or
         rvfi_trap.debug
      or
         rvfi_trap.exception
```

silabs-hfegran, 2022-10-05:
Thank you @silabs-oysteink and @Silabs-ArjanB, I will run a few tests to make sure

closed 2022-10-05 by silabs-hfegran