openhwfoundation/cv32e40s #302: Bits in cpuctrl CSR may be cleared in RMW operations

silabs-oysteink, 2022-09-13:
There is currently no testcase to back up this issue, but inspecting the RTL indicates that the following may be the case:

Any CSR instruction using Read-Modify-Write acting on cpuctrl may clear bits of cpuctrl that should not be cleared.
The reasoning behind this is that the cpuctrl CSR should always read 0. This value is read in EX and piped to WB for usage in the RMW operation and writing to the register file. Since the value read is zero, only bits that will be 1 due to the instruction will end up as ones in the CSR, all other bits will be 0 - including bits that were one before the instruction but that the instruction itself did not touch.

Silabs-ArjanB, 2022-09-13:
Are there other (maybe PMP) CSRs that perform 'read legalization' and therefore have the same issue?

silabs-oysteink, 2022-09-14:
Running the following instructions expose the problem:

1. csrrwi x0, 0xbf0, 0x10 // enable cpuctrl.integrity
2. csrrsi x0, 0bbf0, 0x8   // Set bit for pc_harden

The resulting waveform looks like this:

[image]

Silabs-ArjanB, 2022-09-14:
@silabs-mateilga Can this issue be addressed by a cover point (and/or directed test with the above instructions)?

silabs-mateilga, 2022-09-15:
Both are possible, but I think they will be covered when the autogenerated CSR test is updated, maybe @silabs-hfegran can comment on this?

Apart from exercising the issue, I'm uncertain if the scoreboard would pick it up. That depends on what is reported on rvfi rdata for this special register. Is it the actual stored value, or is it the "read" value 0f 0x0?

silabs-hfegran, 2022-09-15:
This will only be covered by the autogenerated csr test IF the register value is readable - otherwise the test have no way of knowing what value was actually committed to the register on a write.

Silabs-ArjanB, 2022-09-15:
> Apart from exercising the issue, I'm uncertain if the scoreboard would pick it up. That depends on what is reported on rvfi rdata for this special register. Is it the actual stored value, or is it the "read" value 0f 0x0?

We report the 'read value' as opposed to the 'actual stored value' on RVFI. 

@silabs-mateilga What does the ISS actually compare against in its scoreboard? Please see https://github.com/openhwgroup/cv32e40s/issues/255 for an official RISC-V example (pmpaddr) where the read value and the actually stored value can difffer.

silabs-mateilga, 2022-09-15:
As you say the rdata represents the "read value", I see no way ISS can compare with the "actual stored value" as it's not (yet) exposed on the rvfi, and they use that to compare. We'd need actual value exposed to verify this behaviour. Of course, we could do it ourselves by probing the actual register value in the core, but this would be a new addition to the test environment.

Silabs-ArjanB, 2022-09-15:
@silabs-mateilga The question is what value the ISS reports (not what value is reported on RVFI, that we know).

Silabs-ArjanB, 2022-09-26:
@silabs-oysteink cpuctrl needs to be changed to become a regular R/W CSR. The secureseed* CSRs are still 'read as 0', so the reported problem still exists there.

silabs-oysteink, 2023-01-10:
Fix implemented in PR #383 - limiting accesses to CSRRW with rs1 != x0.

closed 2023-04-12 by silabs-oysteink