openhwfoundation/cv32e40s #350: Pending NMI, but going to sleep

silabs-robin, 2022-12-06:
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
1. My branch, [interrupt_assert_fix](https://github.com/silabs-robin/core-v-verif/tree/interrupt_assert_fix) (hash 05a6d7)
2. tcf_cvverif, assert `a_wfi_assert_core_not_ready`
3. (See screenshot below)

There is a pending NMI when the core goes to sleep.

[image]

I don't know if this is a bug or not, because there is no outstanding transactions at the time, and NMI should not(?) be treated as a "locally enabled interrupt".
Note also that `ctrl_fsm_cs == SLEEP` doesn't 1-to-1 mean that the core is in "sleep mode".
The assert in question was still mid-development when the issue arose, so I don't expect the assert itself to be bug-free.

@silabs-oysteink @Silabs-ArjanB

Silabs-ArjanB, 2022-12-06:
@silabs-robin An NMI is not supposed to be covered by the phrase 'locally enabled interrupt'. In itself the above behavior looks okay, but I wonder if we should eliminate this scenario by not entering sleep mode while there is a pending NMI. Now we wait until the LSU is no longer busy and that is one cycle too short to exclude NMIs. @silabs-oysteink let's discuss whether we change the RTL or accept above behavior (I am leaning towards accepting it).

silabs-robin, 2022-12-07:
> An NMI is not supposed to be covered by the phrase 'locally enabled interrupt'

Why not?
I can't find it in any spec or user manual.

Is it not "locally enabled"?
An action cannot be taken to _enable it_, so in that sense you could say it has not **been** "enabled".
But you cannot disable it, so in that sense you could say that it always implicitly **is** "enabled".

Is it not an "interrupt"?
I think it is an interrupt, because it is called "interrupt" and because of all the interrupt-like behaviors around it (e.g. mcause[31]).

Silabs-ArjanB, 2022-12-08:
@silabs-robin Agreed that this is a bug in the RTL and that an NMI shall cause a resume from WFI (and WFE).

Silabs-ArjanB, 2022-12-08:
Can assertions as the following be added as well:

 (core_sleep_o == 1'b1) -> (instr_rvalid_i == 1'b0);

 (core_sleep_o == 1'b1) -> (data_rvalid_i == 1'b0);

 (core_sleep_o == 1'b1) -> 'write buffer is empty'

Silabs-ArjanB, 2022-12-09:
Fixed by https://github.com/openhwgroup/cv32e40x/pull/722 (not merged yet to the 40S)

Silabs-ArjanB, 2022-12-15:
@silabs-robin Please close this issue if you agree it has been solved.

silabs-robin, 2022-12-19:
I will have a look at this and the other issue soon. It is not forgotten.

Silabs-ArjanB, 2023-03-13:
@silabs-robin Can you please close this if it has been fixed?

silabs-robin, 2023-03-13:
The relevant assertions have now turned to "passing".

closed 2023-03-13 by silabs-robin