openhwfoundation/cv32e40s #353: "mtvec.mode" WARL misbehavior

silabs-robin, 2022-12-13:
### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce

Modify "hello-world":
```
int main(int argc, char *argv[])
{
  uint32_t wrval;
  wrval = 0x5a5a5a5a;
  __asm__ volatile( "csrrw x0, mtvec, %0" : : "r"(wrval) );
  return EXIT_SUCCESS;
}
```

Observe ISS miscompare:
`Info (IDV)  0> CSR 305 (mtvec) - dut:5a5a5a00 ref:5a5a5a01`

IMO, this looks like an RTL bug, because:
1. `mtvec.mode` reset value is `2'b 01`
2. We write 0x5a5a5a5a, i.e. attempting `mtvec.mode <- 2'b 10`
3. Unsupported `mtvec.mode`, _WARL (0x0, 0x1)_
4. Should keep previous value, i.e. the reset value `2'b 01`

Note, this is probably a problem on the X as well, it is just that I tested this on the S.

Core hash: dce36a
Cvverif hash: 4fbc68
(Hashes shouldn't be too important. Including just for completeness.)

Silabs-ArjanB, 2022-12-13:
@silabs-oysteink We did not yet implement the resolve functions for all WARL fields. I expect there will be more such cases (will send you a message about it).

Silabs-ArjanB, 2022-12-20:
@silabs-oysteink Hasn't this been fixed? If so add 'resolved' label and add related PR.

silabs-oysteink, 2022-12-20:
Issue should be resolved after PR #735.

silabs-robin, 2023-01-03:
Is fixed.

closed 2023-01-03 by silabs-robin