openhwfoundation/cv32e40s #355: "rvfi_intr" repeated mistakenly

silabs-robin, 2022-12-14:
### Component
**Component:Other**: For any other issues

### Steps to Reproduce

1. cvverif: `cv32e40s/dev`, 48d51f
2. rtl: latest, afa8d7
3. tcf_cvverif, `a_trap_mpp_general`  (that assert is for a different purpose, but caught this)

It seems like there is only 1 interrupt being acked, but some logic in rvfi re-issues the rvfi_intr.
_"// The succeeding instruction therefore needs to re-trigger the intr signals if it it did not reach the rvfi output."_, but the intr _did_ reach the output.

[image]

Silabs-ArjanB, 2022-12-14:
Hi @silabs-robin Could you please add a figure showing the status all rvfi_intr fields?

silabs-robin, 2022-12-14:
> Hi @silabs-robin Could you please add a figure showing the status all rvfi_intr fields?

@Silabs-ArjanB 

[image]

Silabs-ArjanB, 2023-03-13:
@silabs-robin Can you please close this if it has been fixed?

silabs-robin, 2023-03-13:
Hi. I cannot close this atm, because the assert in question has a failure relating to this https://github.com/openhwgroup/cv32e40s/issues/414 .

silabs-oysteink, 2023-03-30:
@silabs-robin Issue #414 should now be resolved, can you check and close if you can confirm?

silabs-robin, 2023-08-31:
Closing based on:
`a_trap_mpp_general` gets a >24 cycle bound (28 so far), rc=se.
Same with `a_intr_count`.
And recent formal cleanup efforts have not reported anything related to this.