openhwfoundation/cv32e40s #361: ISS mismatch for DCSR cause

silabs-halfdan, 2022-12-16:
ISS flags a mismatch after reading the DCSR cause field.

The error happens in a GPR a cycle after the DCSR is read because DCSR checks have been temporarily turned off.

### Steps to Reproduce
Hash: 723c1ba1f5f9f57b4e11c6f2f7640449f1f22f08
Note: Comment out a_dcsr_cause and a_debug_mode_ext_req assertions as they are not ye updated for current RTL
cmd: makeuvmt gen_corev-dv  corev-dv gen_comp comp test TEST=corev_rand_debug

Log :
UVM_INFO @ 114.300 ns : uvmt_cv32e40s_firmware_test.sv(143) uvm_test_top [TEST] Started RUN
Error (IDV) GPR register value mismatch (HartId:0, PC:1a110878):
Info (IDV)  0> GPR x7 - dut:400084d3 ref:40008453
UVM_ERROR @ 318847.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.trace2api.state_compare @ 318847.000 ns: MISMATCH

silabs-halfdan, 2022-12-20:
Root cause is described in #365

Silabs-ArjanB, 2023-01-02:
Should be fixed after merge of https://github.com/openhwgroup/cv32e40x/pull/749 to 40S

closed 2023-01-04 by silabs-halfdan