openhwfoundation/cv32e40s #369: Incorrect linking behavior for cm.jalt and cm.jt

silabs-hfegran, 2022-12-21:
## Incorrect linking behavior for cm.jalt and cm.jt

Confirmed by inspection: RTL appear to 
* not treat `cm.jt` with `index inside {[32:63]}` as `cm.jalt`
* treat `cm.jalt` with `index inside {[32:63]}` as `cm.jt`

This is due to a change in behavior from 0.70.x to 1.0.0-RC5.7, the previous threshold was 64.

[image]

Zc specification v.1.0.0-RC5.7:
> 3.5. cm.jalt
(...)
> * Note: For this encoding to decode as cm.jalt, index>=32, otherwise it decodes as cm.jt, see
Jump via table.

> 3.4. cm.jt
(...)
> * For this encoding to decode as cm.jt, index<32, otherwise it decodes as cm.jalt, see
Jump and link via table.

Silabs-ArjanB, 2022-12-21:
Note that our user manuals are still v1.0.0-RC5.6. I will soon release 0.7.0 versions that will also point to v1.0.0-RC5.7 amongst other changes. RTL updates will need to be aligned with updates to ISS and binutils.

Silabs-ArjanB, 2023-01-16:
Solved with https://github.com/openhwgroup/cv32e40x/pull/763 (not merged to 40S yet)

Silabs-ArjanB, 2023-03-13:
@silabs-hfegran Can you please close this if it has been fixed?

closed 2023-03-14 by silabs-hfegran