openhwfoundation/cv32e40s #384: Non-completed Instructions Get rvfi_valid

silabs-robin, 2023-01-10:
### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)
**Component:Other**: For any other issues

### Steps to Reproduce
Please provide:
1. On my branch of cvverif, 102757,  https://github.com/silabs-robin/core-v-verif/tree/interrupt_assert_fix2_recarpet_reboot
2. Use latest RTL, 467acc
3. tcf_cvverif, `a_wfi_assert_come_coresleepo`

### Problem

Trigger match has "before" timing, and so the instruction should not complete (not retire, not trap, not commited, not anything), but it ends up on rvfi.

[image]

This specific trace and assert is about wfi, but the principle is wrong in general.
And if this behavior is there for historical reasons to please the iss, then I don't think that is a good enough reason to be wrong and that a new solution (if necessary) should be found to please the iss.

Silabs-ArjanB, 2023-01-10:
I don't see the issue: RVFI valid is allowed/expected to be asserted on non-retired instructions (i.e. with rvfi_trap = 1) and we defined trigger matches to also set rvfi_trap = 1.

silabs-robin, 2023-01-10:
Hopefully I gathered my thoughts on this correctly. But I don't want to spend too much time so I'm just hitting send without proof reading comprehensively.

Here is what I was thinking:
"Before" timing is defined as, _"The action for this trigger will be taken just *before* the instruction that triggered it is committed"_.
Therefore, the instruction doesn't run and cannot trap, and doesn't fulfil _"Any RVFI operation (retired or trapped instruction) will set rvfi_valid"_. (Imagine that the instruction would have caused an illegal instruction exception, then that illegal instruction exception could not occur!)

Here is an _almost_ satisfactory answer to why it is wrong:
Firstly, _"execution attempt"_ is not the same as _"commited instruction"_, and a trigger match must do the former but not the latter (being in wb, non-halted, non-killed).
Secondly, riscv _"exceptions"_ are defined quite generally, as _"an unusual condition occurring at run time associated with an instruction in the current RISC-V hart"_, so it merely requires being "associated with" an instruction and doesn't require "committing" it.
Hence, the non-commited instruction can cause an "exception" and so it fulfils the rvfi_valid criteria, although this "exception" is of a different kind than the mcause exceptions.

_(Which seems to mean there is a difference between {normal exception, debug exception, normal exception causing debug} so that single stepping is the only case that can have a "true" exception causing debug entry, and the debug cause table could say "[debug] trap code" instead of "cause encoding", and the mcause table should say "trap code" instead of "exception code".)_

closed 2023-01-11 by silabs-robin