openhwfoundation/cv32e40s #388: Wrong clearing of bits in pmp_addr_q

silabs-oysteink, 2023-01-12:
## Wrong clearing of bits in pmp_addr_w

### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce

1. RTL hash 5a0df34380e0b885cac439c4b2b3d2ea75f3f625
2. Running formal verification, the assertion "a_set_clear_pmp_addr_q" within cv32e40s_cs_registers_sva.sv should fail.

When granularity >=1 and the mode is either OFF or TOR, the read data resolution will tie off the least significant implemented (flipflop) bit to 0. This may propagate back through the RMW operation and inadvertently clear the flipflop when it shouldn't.

silabs-oysteink, 2023-01-12:
This is actually in reality a duplicate of issue #302. Keeping it separate as it is very PMP specific. When fixed, both this issue and #302 may likely be closed.

Silabs-ArjanB, 2023-04-12:
@silabs-oysteink Can this issue and https://github.com/openhwgroup/cv32e40s/issues/302 both be (fully) closed now?

silabs-oysteink, 2023-04-12:
Yes, I will close them.

closed 2023-04-12 by silabs-oysteink