openhwfoundation/cv32e40s #397: Tselect side effects not reported on rvfi

silabs-halfdan, 2023-01-26:
**Component:RTL** - rvfi

Side-effects of writes to tselect are not attributed to the csr instruction that caused them.

Writing a new value to the tselect register should result in updated values for the tdata csr rdata output signals on rvfi.
Currently this updated register value is not attributed to the correct instruction on rvfi.

## Steps to Reproduce
Configure the core to have multiple triggers.
Write tdata1 to non-reset value, then write 1 to tselect. This should have resulted in side effects like updated value of the tdata1 csr outputs in rvfi.
###Specifically
1. Use this branch: https://github.com/silabs-halfdan/core-v-verif/tree/rvfi_tselect_side_effect_issue
2. Run 'makeuvmt test TEST=debug_test_trigger CFG=debug_trigger_cfg2 GUI=yes ADV_DEBUG=yes WAVES=yes'

## Analysis

In this case, instruction 14a should be reported on rvfi with the related side-effect; show tdata1[1] instead of tdata1[0].

Object dump:
```
    146:	f281a903          	lw	x18,-216(x3) # 128e8 <trigger_sel>
    14a:	7a091073          	csrrw	x0,tselect,x18
```
Waves:
[image]

Silabs-ArjanB, 2023-01-26:
>> Writing a new value to the tselect register should result in updated values for the tdata csr rdata output signals on rvfi.

tdata_csr_rdata reflects the read data as seen by the instruction itself, so the 'before state'. So the instruction after this csrrw should show the tdata_csr_rdata that you are interested in, so it sounds like this RVFI signal is not being interpreted correctly in the verification environment.

You could argue that we should report the tdata side effect via its csr_tdata_wdata field (and corresponding mask field).

silabs-halfdan, 2023-01-26:
Good point, looks like we're expecting the the before state of the wrong register because we're checking it using the tselect from wdata.
Ideally, we would see the side-effects on wdata, but for now I think it is just a nice-to-have. I will close this ticket and open one on core-v-verif.

Silabs-ArjanB, 2023-01-26:
@silabs-halfdan I just wanted to highlight that the expectations on rdata seem wrong; I am fine however if you keep this ticket open if you want us to report these side effects via wdata instead.

silabs-halfdan, 2023-01-30:
Re-opening to get side-effects reported on rvfi wdata after local discussions.

Silabs-ArjanB, 2023-02-01:
Should be fixed by https://github.com/openhwgroup/cv32e40x/pull/773 (merge into 40S to be done)

silabs-halfdan, 2023-02-01:
Will close when #401 is merged.

Silabs-ArjanB, 2023-02-01:
Merged now

closed 2023-02-01 by silabs-halfdan