openhwfoundation/cv32e40s #403: WFE/WFI in wb + pending irq can cause interrupt to be missed

silabs-hfegran, 2023-02-06:
## WFE/WFI in wb + IRQ can cause interrupt to be missed
An wfi/wfe in wb while there is a pending and enabled interrupt for one cycle causes interrupt to be missed. In the example below the second pending and enabled interrupt wakes up the core and is taken.

### Steps to Reproduce
1. git hash: df6c2cd793b4
2. Assertion:
```
    property p_higher_lvl_than_mintthresh_th_can_preempt;
            clic.irq
        ##1 ($past(clic.priv) == current_priv_mode)
        &&  ($past(clic.level) > effective_clic_level)
        &&  mstatus_fields.mie
        &&  is_interrupt_allowed == 1'b1
      |->
            irq_ack
      or
            rvfi_valid[->1:2]
        ##0 rvfi_dbg_mode
      or
            rvfi_valid[->1:2]
        ##0 rvfi_intr.exception
      or
            rvfi_valid[->1:2]
        ##0 rvfi_trap.exception
      or
            rvfi_valid[->1:2]
        ##0 is_cause_nmi
      ;
    endproperty : p_higher_lvl_than_mintthresh_th_can_preempt

    a_higher_lvl_than_mintthresh_th_can_preempt: assert property(p_higher_lvl_than_mintthresh_th_can_preempt)
    else
      `uvm_error(info_tag,
        $sformatf("Higher level than mintthresh should be able to interrupt"));

```
3. Logfile and/or wave-dump info (screen shots can be useful)
[image]

Silabs-ArjanB, 2023-02-13:
Not an RTL bug, but the interrupt_allowed signal should be fixed nonetheless to have a more logical behavior for 'multi-op' WFE/WFI. Should be SEC clean, but will aid writing assertions.

Silabs-ArjanB, 2023-02-28:
Resolved with https://github.com/openhwgroup/cv32e40x/pull/795

silabs-oysteink, 2023-02-28:
The merge from CV32E40X will come shortly.

Silabs-ArjanB, 2023-03-13:
@silabs-hfegran Can you please close this if it has been fixed?

silabs-hfegran, 2023-03-13:
@Silabs-ArjanB Will close this issue when verified

silabs-oysteink, 2023-04-13:
Is this verified yet?