openhwfoundation/cv32e40s #410: RVFI misrepresents OBI transactions?

silabs-robin, 2023-02-27:
### Component
**Component:Other**: For any other issues

### Steps to Reproduce
1. Cvverif hash `f7f2d5`
2. My fork https://github.com/silabs-robin/core-v-verif
3. tcf_cvverif
4. a_obi_vs_rvfi

If I there is nothing I have overlooked, it seems like `rvfi_mem_*` reports memory writes that didn't happen.
In the example below it seems like there is an MPU error, so the bus access is denied, but RVFI says the mem got updated.

[image]

Silabs-ArjanB, 2023-02-27:
It looks like an RVFI bug indeed (it ignores the related valid signal and acts as if a memory write actually happened).

@silabs-robin I would assume that rvfi_trap.trap gets asserted as well, right?

silabs-robin, 2023-02-27:
> I would assume that rvfi_trap.trap gets asserted as well, right?

Yes.

[image]

Silabs-ArjanB, 2023-03-27:
Should be fixed by https://github.com/openhwgroup/cv32e40x/pull/816 (once merged from 40X to 40S)

silabs-robin, 2023-04-11:
I am seeing a bound of more than double where the original CEX was found, and we have reports of a bound of "26 cycles / 12k seconds ".
Therefore, I consider the problem fixed.

_(If there exists any further problem on deeper depths then I will create a new issue for that separately.)_

closed 2023-04-11 by silabs-robin