openhwfoundation/cv32e40s #412: Mismatch between manual and RTL for tdata2 (for tdata1.type==2)

silabs-halfdan, 2023-03-02:
# Mismatch between RTL and reference manual

In the [reference manual](https://cv32e40s-user-manual.readthedocs.io/en/latest/control_status_registers.html#trigger-data-register-2-tdata2-view-when-tdata1-type-is-0x2) the register is described as 32 bits of **DATA** which implies any value is legal. In the [RTL](https://github.com/openhwgroup/cv32e40s/blob/e9f81d2193f3b1c52733ac20105dc5e1352c0666/rtl/cv32e40s_debug_triggers.sv#LL243C59-L243C59) however only bits that correspond to valid exception codes are writable.

This triggers an ISS mismatch as the ISS implements the register with all bits writeable as implied in the manual.

Silabs-ArjanB, 2023-03-02:
Hi @silabs-halfdan I think that the RTL and reference manual match for tdata2 with tdata1.type == 5, see https://cv32e40s-user-manual.readthedocs.io/en/latest/control_status_registers.html#trigger-data-register-2-tdata2-view-when-tdata1-type-is-0x5. The tdata2 register is not described as 32 bits of RW data as you imply above.

Should this issue therefore be rephrase to 'mismatch between ISS and reference manual / RTL'?

silabs-halfdan, 2023-03-03:
@Silabs-ArjanB, yes I looked at the register description for the wrong tdata1.type... I will update the title and description.

silabs-halfdan, 2023-03-03:
I will close the issue here, and open one on core-v-verif.

closed 2023-03-03 by silabs-halfdan