openhwfoundation/cv32e40s #414: RVFI - Exception Handler Without Trap

silabs-robin, 2023-03-09:
### Component
**Component:Other**: For any other issues

### Steps to Reproduce
1. Cvverif fork: https://github.com/silabs-robin/core-v-verif/tree/umode_assert_new
2. Hash: `1a63e9`
3. `tcf_cvverif`
4. Assert: `rvfi_assert_i.a_sync_cause`

Problem:
(See screenshot below.)
There are 2 `rvfi_valid`, where the second indicates `rvfi_intr.exception`, but the first one didn't even have `rvfi_trap.trap`.
This should not be possible.

[image]

### Additional Details

Seems like there are 3 `wb_valid`, but only 2 `rvfi_valid`:
[image]

The middle `wb_valid` actually has an OBI `err`, matching the `rvfi_intr.cause=24` ("Instruction bus fault").
[image]

So it seems like the middle `wb_valid` was mistakenly suppressed from reaching `rvfi_valid`.
_(Otherwise it would have reported "rvfi_trap.exception", so the following "rvfi_intr.exception" would be ok.)_

silabs-oysteink, 2023-03-10:
This is expected behaviour for RVFI as currently only instructions can cause rvfi_valid. rvfi_valid is set for retired instructions and instructions causing exceptions or debug entry via ebreak or trigger match. In the traces above, there is a clic pointer which finishes in WB, but as a pointer isn't an instruction it does not cause rvfi_valid. There is definitely a 'whole' in the RVFI here, in that the pointer does not get reported. However, if we were to add pointers to rvfi_valid I presume a lot of other infrastructure that uses RVFI would need an update as well. Perhaps some sort of side channel could be used instead, as to not disturb current code that uses RVFI.

Silabs-ArjanB, 2023-03-10:
@silabs-oysteink Would it be easy to at least report this with a specific rvfi_trap type?

silabs-oysteink, 2023-03-10:
We would need to capture the exception for the pointer inside RVFI, and then on the next rvfi_valid we would somehow need to signal the special rvfi_trap. It shouldn't be too hard I believe.

silabs-robin, 2023-03-10:
> In the traces above, there is a clic pointer which finishes in WB, but as a pointer isn't an instruction it does not cause rvfi_valid.

Why is it not treated as an instruction? Is it not part of the fetch of an instruction? I don't see how it is fundamentally different from a regular instruction fetch that goes wrong, except that it is involves an extra step.

Is there specifically any spec that says that it has to be the way that we currently do it?

silabs-mateilga, 2023-03-10:
I just want to note that I see the same behaviour as problematic with regards to debug asserts. I have to weaken some of the asserts by waving certain conditions to work around this, or put a lot of effort in to recognising these special cases. I also agree with Robin that there is no conceptual difference to any other failed instruction fetch here, which are already reported on rvfi as trapped "instructions".

Silabs-ArjanB, 2023-03-13:
@silabs-robin 

>> Why is it not treated as an instruction? 

It is a pointer, not an instruction.

>> Is it not part of the fetch of an instruction? 

No, say the CLIC pointer table entry is at address 0x1000 and the actual instruction is at address 0x20000. In this case the exception relates to address 0x1000. If the error was related to 0x20000 we would have attributed it to the instruction. Similarly if the this was not a CLIC pointer but a table jump then a fault related to address 0x1000 would be attributed to the table jump and not to the target instruction. In case of an interrupt there is simply nothing to attribute the error to.

>> I don't see how it is fundamentally different from a regular instruction fetch that goes wrong, except that it is involves an extra step. 

The extra step is not related to addres 0x20000.

>> (It seems analogous to how a "split" misaligned access would give an exception if either of those two fetches has an error.)

In that case both accesses clearly relate to the given instruction.

>> Is there specifically any spec that says that it has to be the way that we currently do it?

The user manual states that we signal rvfi_valid for retired or trapped _instructions_.

If we start reporting rvfi_valid more often than surely a ton of assertions and integration with the ISS will break down and will need fixing.

We could easily add an additional field to rvfi_intr, for example rvfi_intr.clicptr, to indicate that a rvfi_intr.exception = 1 really relates to the clic_ptr. Would that help you? Would it be better if we would set rvfi_intr.clicptr for any rvfi_intr due to a CLIC pointer (not only those due to exceptions)?

Alternatively we could change rvfi_valid as you want, but that sounds like a lot more work especially on the verification side. Should we do then do a similar thing for the similar part of a table jump, etc.? We prefer not to go here.

silabs-robin, 2023-03-14:
> Why is it not treated as an instruction?
>
> It is a pointer, not an instruction.

It is a pointer, and therefore it can not be treated as being part of an instruction?
This cannot be true, because as you said yourself table jump pointers are "attributed" to the table jump instruction.

--------

> Similarly if the this was not a CLIC pointer but a table jump then a fault related to address 0x1000 would be attributed to the table jump and not to the target instruction

Yes, and that is how table jumps work.
There is nothing yet that says that this must translate exactly to how the CLIC vector table works.

--------

Please answer my other comment as well.
https://github.com/openhwgroup/cv32e40s/issues/414#issuecomment-1463730892
This is the core of the problem.

Put differently: If `rvfi_intr.exception` indicates that the trap is from a synchronous exception, what is the failed clic pointer fetch synchronous to?

Silabs-ArjanB, 2023-03-29:
Should be fixed with https://github.com/openhwgroup/cv32e40x/pull/817 (after that is merged to 40S).

silabs-oysteink, 2023-03-30:
@silabs-robin Issue should now be resolved, can you check and close if you can confirm?

silabs-robin, 2023-04-11:
The assert's proof attempt now progresses sufficiently far past the original issue (20 now, 16 before).
Considering this issue fixed.
(If there any future problems are revealed it shall be considered a new issue.)

closed 2023-04-11 by silabs-robin