openhwfoundation/cv32e40s #415: Signal debug_pc_o shows pointer address with clic and mret

silabs-mateilga, 2023-03-10:
**Component:RTL**

When a mret instruction restarts a clic pointer fetch, the expected value for debug_pc_o is the address of the mret instruction, but it actually reports the address of the pointer. See pipeline vizualisation in the attached image.

Reproduce by formally proving assert a_debug_pc_o_inv in the debug_asserts assertion set.

[image]

Silabs-ArjanB, 2023-03-16:
Still needs merge from 40X

silabs-oivind, 2023-03-20:
@silabs-mateilga . This is merged to e40s now. Please close ticket if you agree it's fixed.

silabs-mateilga, 2023-03-23:
Fix works as intended, assert formally proven

closed 2023-03-23 by silabs-mateilga