openhwfoundation/cv32e40s #419: Clarification of trigger register implementation for 0 triggers

silabs-halfdan, 2023-03-13:
In the debug spec, implementing the trigger registers is optional if no triggers are implemented.
The reference manual does not clearly state whether or not the registers are implemented (with WARL(0x0)) when no triggers are implemented.

This can lead to misunderstandings.
One example is the ISS implementation of tcontrol, where the reference manual states the CSR is hardwired to 0 but does not state whether it exists for DBG_NUM_TRIGGERS==0. So for ISS it is always implemented and hardwired to 0 (addressed in separate issue).

Silabs-ArjanB, 2023-03-13:
Hi @silabs-halfdan Doesn't the following clarify it good enough?

[image]

silabs-halfdan, 2023-03-14:
@Silabs-ArjanB I also filed as an ISS bug. I guess this is explicit enough, as this did not seem to be the reason for the ISS implementation. Closing this ticket.

closed 2023-03-14 by silabs-halfdan