openhwfoundation/cv32e40s #429: tdata2 does not resolve to legal value

silabs-hfegran, 2023-03-30:
## tdata2 does not resolve to legal value
Write all Fs to tdata2 with type 5 triggers set causes bits 6 and 4 to be set (should be WARL 0x0) 
[image]

### Steps to Reproduce
Please provide:
1. core-v-verif, cv32e40s/dev, hash: `460b5c9c8c0d69d4247ab18ba3a06cfa0114727b` 
2. make test TEST=debug_test_trigger CFG=debug_trigger_cfg2 USE_ISS=YES COV=YES RUN_INDEX=6 GEN_START_INDEX=6 RNDSEED=327595069 
3. Logs:
```diff
false, true, 0, NONE, NONE, 24,183,300,    24183.000 ns | RVFI |     8055 |   5358 | 1a11097a | 7a105073 | M | x0  | 00000000 | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - csrrwi x0,tdata1,0
false, true, 0, NONE, NONE, 24,186,300,    24186.000 ns | RVFI |     8056 |   5359 | 1a11097e | fff00493 | M | x0  | 00000000 | x0  | 00000000 | x9  | ffffffff |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - addi x9,x0,-1
false, true, 0, NONE, NONE, 24,192,300,    24192.000 ns | RVFI |     8058 |   5360 | 1a110982 | 7a249073 | M | x9  | ffffffff | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - csrrw x0,tdata2,x9
false, true, 0, NONE, NONE, 24,195,300,    24195.000 ns | RVFI |     8059 |   5361 | 1a110986 | 60000437 | M | x0  | 00000000 | x0  | 00000000 | x8  | 60000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - lui x8,0x60000
false, true, 0, NONE, NONE, 24,198,300,    24198.000 ns | RVFI |     8060 |   5362 | 1a11098a | ----147d | M | x8  | 60000000 | x0  | 00000000 | x8  | 5fffffff |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - c.addi x8,-1 # 5fffffff <load_store_test_high_addr+0x45eeeffb>
false, true, 0, NONE, NONE, 24,201,300,    24201.000 ns | RVFI |     8061 |   5363 | 1a11098c | 7a141073 | M | x8  | 5fffffff | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - csrrw x0,tdata1,x8
false, true, 0, NONE, NONE, 24,210,300,    24210.000 ns | RVFI |     8064 |   5364 | 1a110990 | 7a1024f3 | M | x0  | 00000000 | x0  | 00000000 | x9  | f8000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - csrrs x9,tdata1,x0
false, true, 0, NONE, NONE, 24,213,300,    24213.000 ns | RVFI |     8065 |   5365 | 1a110994 | f8000437 | M | x0  | 00000000 | x0  | 00000000 | x8  | f8000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - lui x8,0xf8000
false, true, 0, NONE, NONE, 24,219,300,    24219.000 ns | RVFI |     8067 |   5366 | 1a110998 | 00940563 | M | x8  | f8000000 | x9  | f8000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - beq x8,x9,1a1109a2 <_debug_mode_register_test+0x11c>
false, true, 0, NONE, NONE, 24,228,300,    24228.000 ns | RVFI |     8070 |   5367 | 1a1109a2 | 7a205073 | M | x0  | 00000000 | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1383 - csrrwi x0,tdata2,0
false, true, 0, NONE, NONE, 24,231,300,    24231.000 ns | RVFI |     8071 |   5368 | 1a1109a6 | 60000437 | M | x0  | 00000000 | x0  | 00000000 | x8  | 60000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - lui x8,0x60000
false, true, 0, NONE, NONE, 24,234,300,    24234.000 ns | RVFI |     8072 |   5369 | 1a1109aa | ----147d | M | x8  | 60000000 | x0  | 00000000 | x8  | 5fffffff |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - c.addi x8,-1 # 5fffffff <load_store_test_high_addr+0x45eeeffb>
false, true, 0, NONE, NONE, 24,237,300,    24237.000 ns | RVFI |     8073 |   5370 | 1a1109ac | 7a141073 | M | x8  | 5fffffff | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - csrrw x0,tdata1,x8
false, true, 0, NONE, NONE, 24,246,300,    24246.000 ns | RVFI |     8076 |   5371 | 1a1109b0 | 7a1024f3 | M | x0  | 00000000 | x0  | 00000000 | x9  | 58000241 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - csrrs x9,tdata1,x0
false, true, 0, NONE, NONE, 24,249,300,    24249.000 ns | RVFI |     8077 |   5372 | 1a1109b4 | 58000437 | M | x0  | 00000000 | x0  | 00000000 | x8  | 58000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - lui x8,0x58000
false, true, 0, NONE, NONE, 24,252,300,    24252.000 ns | RVFI |     8078 |   5373 | 1a1109b8 | 24140413 | M | x8  | 58000000 | x0  | 00000000 | x8  | 58000241 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - addi x8,x8,577 # 58000241 <load_store_test_high_addr+0x3deef23d>
false, true, 0, NONE, NONE, 24,258,300,    24258.000 ns | RVFI |     8080 |   5374 | 1a1109bc | 00940563 | M | x8  | 58000241 | x9  | 58000241 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - beq x8,x9,1a1109c6 <_debug_mode_register_test+0x140>
false, true, 0, NONE, NONE, 24,264,300,    24264.000 ns | RVFI |     8082 |   5375 | 1a1109c6 | ----0001 | M | x0  | 00000000 | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1397 - c.addi x0,0
+false, true, 0, NONE, NONE, 24,267,300,    24267.000 ns | RVFI |     8083 |   5376 | 1a1109c8 | 50000437 | M | x0  | 00000000 | x0  | 00000000 | x8  | 50000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - lui x8,0x50000
+false, true, 0, NONE, NONE, 24,270,300,    24270.000 ns | RVFI |     8084 |   5377 | 1a1109cc | 7a141073 | M | x8  | 50000000 | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - csrrw x0,tdata1,x8
+false, true, 0, NONE, NONE, 24,279,300,    24279.000 ns | RVFI |     8087 |   5378 | 1a1109d0 | 7a1024f3 | M | x0  | 00000000 | x0  | 00000000 | x9  | 58000001 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - csrrs x9,tdata1,x0
false, true, 0, NONE, NONE, 24,282,300,    24282.000 ns | RVFI |     8088 |   5379 | 1a1109d4 | 58000437 | M | x0  | 00000000 | x0  | 00000000 | x8  | 58000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - lui x8,0x58000
false, true, 0, NONE, NONE, 24,285,300,    24285.000 ns | RVFI |     8089 |   5380 | 1a1109d8 | ----0405 | M | x8  | 58000000 | x0  | 00000000 | x8  | 58000001 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - c.addi x8,1 # 58000001 <load_store_test_high_addr+0x3deeeffd>
false, true, 0, NONE, NONE, 24,291,300,    24291.000 ns | RVFI |     8091 |   5381 | 1a1109da | 00940563 | M | x8  | 58000001 | x9  | 58000001 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - beq x8,x9,1a1109e4 <_debug_mode_register_test+0x15e>
false, true, 0, NONE, NONE, 24,297,300,    24297.000 ns | RVFI |     8093 |   5382 | 1a1109e4 | ----0001 | M | x0  | 00000000 | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1408 - c.addi x0,0
-false, true, 0, NONE, NONE, 24,300,300,    24300.000 ns | RVFI |     8094 |   5383 | 1a1109e6 | fff00493 | M | x0  | 00000000 | x0  | 00000000 | x9  | ffffffff |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1420 - addi x9,x0,-1
-false, true, 0, NONE, NONE, 24,303,300,    24303.000 ns | RVFI |     8095 |   5384 | 1a1109ea | 7a249073 | M | x9  | ffffffff | x0  | 00000000 | x0  | 00000000 |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1420 - csrrw x0,tdata2,x9
-false, true, 0, NONE, NONE, 24,312,300,    24312.000 ns | RVFI |     8098 |   5385 | 1a1109ee | 7a202473 | M | x0  | 00000000 | x0  | 00000000 | x8  | 030009fe |  -- | -------- | -------- | _debug_mode_register_test debug_test_trigger.c 1420 - csrrs x8,tdata2,x0

```
[image]

silabs-oysteink, 2023-03-31:
The exception codes 4 and 6 are used on cv32e40x for misaligned atomics. The mask for legal tdata2 values likely got these codes in via the last merge, I will check and make a fix.

Silabs-ArjanB, 2023-03-31:
Hi @silabs-oysteink could you please do a separate user manual update for only the 40x documentation as well as that should mark these bits as warl instead of warl 0.

silabs-oysteink, 2023-03-31:
> Hi @silabs-oysteink could you please do a separate user manual update for only the 40x documentation as well as that should mark these bits as warl instead of warl 0.

Yes, I just reviewed it myself as well based on this issue, and the PR will come soon.

silabs-oysteink, 2023-04-13:
@silabs-hfegran Can this issue be closed now?

closed 2023-04-13 by silabs-hfegran