openhwfoundation/cv32e40s #432: behavior of NMI in CLIC mode

eroom1966, 2023-04-03:
@Silabs-ArjanB @silabs-hfegran 
Can I ask how is NMI handled in CLIC mode ?
I see nothing that documents this, it appears as though it is routed through the CLIC interface somehow

Silabs-ArjanB, 2023-04-03:
Hi @eroom1966 It is described at https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/exceptions_interrupts.html#non-maskable-interrupts. Anyhting specific you are wondering about that is not documented there?

eroom1966, 2023-04-03:
This does not describe how mcause is updated in CLIC mode.

Silabs-ArjanB, 2023-04-03:
Is the following not clear enough?

[image]

eroom1966, 2023-04-03:
I am searching for this section, is this part of 0.8.0, or is this later ?
(**update**, this is not in the 0.8.0 spec)

BTW, I think minsttatus is a typo ?

Silabs-ArjanB, 2023-04-03:
@eroom1966 I will correct the typo. This is the most recent user manual indeed (link provided in my initial response). The mintstatus remark was added because of a recent clarification to the RISC-V CLIC specification.

eroom1966, 2023-04-03:
(@silabs-hfegran copy)
I just want to confirm that upon an NMI being taken, mintstatus should **NOT** change value, as you have described above.

We are observing that mintstatus is in fact cleared to zero by the RTL when taking an NMI.

Is this incorrect behavior in the RTL, or incorrect in the spec?

Can the specification be made more specific about the behavior of NMI in CLIC and non-CLIC modes? Currently, it just says:

> _Non Maskable Interrupts (NMIs) update mepc, mcause and mstatus **similar to regular interrupts**._ 

We have seen that when an NMI is taken in CLIC mode, mcause is updated in a different way to non-CLIC mode (the CLIC-mode mcause.pil field is modified as if this was a CLIC interrupt). Because NMI behavior is custom, I think the specification should explicitly state what happens and precisely what state is modified in all modes.

eroom1966, 2023-04-03:
The latest (under cre-v-verif) is

eroom1966, 2023-04-03:
Looks like this is the RTL under use in core-v-verif

commit a9c6c716e4d1a219775b968bafec239f1190ac16 (HEAD)
Merge: 189417e5 44a9ae66
Author: Arjan Bink <40633348+Silabs-ArjanB@users.noreply.github.com>
Date:   Thu Feb 23 11:18:33 2023 +0100

    Merge pull request #409 from silabs-oysteink/silabs-oysteink_merge-w8-2
    
    Merge from CV32E40X

Silabs-ArjanB, 2023-04-03:
Hi @eroom1966 That RTL still had the bug (i.e. clearing mintstatus). Current RTL should not affect mintstatus due to NMIs.

Silabs-ArjanB, 2023-04-28:
Hi @eroom1966 Can this issue be closed now? If not, what info is missing?

Silabs-ArjanB, 2023-05-23:
Hi @eroom1966 Closing this issue due to lack of response. If there are still questions related to this please open a new issue.

closed 2023-05-23 by Silabs-ArjanB