openhwfoundation/cv32e40s #464: Unwarranted PMP Fault?

silabs-robin, 2023-05-26:
### Component
**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

### Steps to Reproduce
1. My core-v-verif fork/branch: https://github.com/silabs-robin/core-v-verif/tree/pmprvfi_fix3
1. `668b73952c4186934165922614c862ff61de0473`
1. Run formal
1. Check `a_rvfi_mem_allowed_generalinstr` (Bound 14, `-engine_mode B`)

### Description
Two `rvfi_valid`s occur, the latter with a PMP fault exception.
But, investigating the signals show that the instr-side PMP check for both fetches seems fine.

[image]

silabs-robin, 2023-05-30:
@silabs-oivind ref discussion last Friday.

silabs-oivind, 2023-05-30:
Hi,
This is a table jump, where the table jump instruction passes PMP checks, but the table jump target address fails PMP checks.
It seems like the design and RVFI behaves according to spec.

I'm not sure if you have enough information on rvfi to recognize and properly check this scenario. We might need to add some extra rvfi signals for this.

silabs-robin, 2023-05-30:
Understood. I will update the assert accordingly, and request additional RVFI signals if needed.
Closing this issue as it seems the RTL/RVFI is doing the right thing.

closed 2023-05-30 by silabs-robin