openhwfoundation/cv32e40s #524: Instructions being skipped by Prefetcher

ChristianLarmann, 2024-01-23:
Git hash: 0494765
Affected RTL-modules: if_stage_i and submodules (as prefetch_unit_i and instruction_obi_i)

### Problem description

I encountered an issue that sometimes the Prefetch Unit does not process some instructions although they are read from memory and provided by the OBI Instruction interface. I figured out that it does not occur anymore if I set MAX_OUTSTANDING = 1 (instead of =2) in the 'if_stage_i' which is further used in 'alignment_buffer_i' and 'instruction_obi_i'.  

Here an example:

This is the code, which should get the instructions from 8080, then jump to 80c8, and then 8090 etc.

8080: 00c0006f j 0x808C
(8084: 0d00006f j 0x8154)
(8088: 0e40006f j 0x816C)
808c: 30501073 csrw mtvec,zero
8090: 00000093 li ra,0
8094: 00008113 mv sp,ra 
...

And this is what happens with  MAX_OUTSTANDING = 2. As you can see, after around 140 ns the instruction 30501073 arrives i resp_i but it does not become valid. Instead the next instruction 00000093 is assigned to the pc=808C, as can be seen later in the if_id_pipe. For me, it seems that when instr_cnt_q is not 0, the prefetch unit ignores incoming instructions. For me MAX_OUTSTANDING=1 fixed it but I am not sure if it might have side effects that I will regret later on. 

[image]
(Setup: I use Vivado for simulation and connected the core to a cache that connects to BRAM. Boot_address=0x8080)

So my question is, is this behavior intended and why does it not work for me without modifications? Or is there something fundamentally wrong in how I integrated the core?

Exported waveform: [waveform_prefetcher_issue.zip](https://github.com/openhwgroup/cv32e40s/files/14024401/waveform_prefetcher_issue.zip)

silabs-oysteink, 2024-01-23:
Hi Christian,

To me this looks like an issue outside of the cv32e40s itself. I opened your .vcd file and attached an annotated image:

[image]

There are three annotated request/response pairs:

1. First request, address=0x8080 expected response is correct (0x00c0006f)
2. Second request, address=0x8084, expected response is **incorrect** (0x30501073 which corresponds to address 0x808C instead of 0x8084)
3. Third request, address=0x808C (this is the jump target fetch due to the first instruction). Response is **incorrect** (0x00000093 which corresponds to address 0x8090 instead of 0x808C)

The response from the second request arrives after the first instruction has requested a new address to fetch due to the jump. This response has to complete per OBI protocol although it will not be used by the core, and thus the response for 2) is flushed by the prefetcher when it arrives.

It looks to me like your memory system outside of the cv32e40s does not correctly handle multiple outstanding transactions, the core sends correct request addresses to the instruction OBI but gets wrong instructions in return.

ChristianLarmann, 2024-01-23:
I see, it totally makes sense. So the MAX_OUTSTANDING has to be matched with how many multiple outstanding transaction the outside memory system supports. In my case it is only 1 then. 

Thank you the clarification!

silabs-oysteink, 2024-01-23:
As the MAX_OUTSTANDING is a localparam it has not been properly tested with values other than 2. I would suggest keeping it at 2 and instead debug the memory system outside of the core to avoid possible issues related to changing the localparam.

-Øystein-

closed 2024-01-29 by Silabs-ArjanB