openhwfoundation/cv32e40s #530: Instruction Fetch Requests Generate X's during Fuzz Testing

cnokes14, 2025-01-24:
## Instruction Fetch Requests Generate X's during Fuzz Testing
During fuzz testing, instruction requests (instr_req_o and instr_reqpar_o) can have X's.

This seems to be because of a cyclic path of unclocked signals across stages:
lsu_ready_i (execute stage) controls wb_ready_o (execute stage)
wb_ready_o (execute stage) controls wb_ready_i (controller)
wb_ready_i (controller) controls lsu_ready_o (controller)
lsu_ready_o (controller) controls ready_0_i (load/store stage)
ready_0_i (load/store stage) controls ready_0_o (load/store stage)
ready_0_o (load/store stage) controls lsu_ready_i (execute stage)

In cases where instr_req_o is not X, it is because this path is cancelled out (IE: OR'd with 1, AND'd with 0) by additional logic somewhere along the way.

Additionally, if the PMP is enabled and configured, the X may occur during a cycle where instr_addr_o should not be accessible by the core (EX: if MMWP is set, the core may set instr_req_o = X while instr_addr_o is not covered by the PMP).

### Component
Issue seems to span multiple RTL components, namely cv32e40s_ex_stage.sv, cv32e40s_controller.sv, cv32e40s_load_store_unit.sv, and cv32e40s_core.sv.

### Steps to Reproduce
1. Use [this fuzz test](https://github.com/cnokes14/CV32E40S-Fuzz-Integrity-Checking-TB/blob/main/cv32e40s_fuzz_tb.sv).
2. X's seem to occur relatively frequently.

[image]
This screenshot is from running the test I provided. Of note, the X that occurs at 185ns occurs alongside an instruction address which should be blocked by the PMP.

Silabs-ArjanB, 2025-01-24:
Hi @cnokes14 , thank you for your detailed report. The cyclic path you report cannot be as stated as the execute stage does not control wb_ready_o (wb_ready comes from the WB stage), so the following parts in bold cannot be true:

lsu_ready_i (execute stage) controls **wb_ready_o (execute stage)**
**wb_ready_o (execute stage)** controls wb_ready_i (controller)

If there is a loop it must be a different one; could you please correct above loop info?

Are you 100% sure all inputs to the top level are non-X at all times (other then maybe time 0)? (I cannot see that from above screenshot)

Best regards,
Arjan

cnokes14, 2025-01-24:
Hi Arjan,

I'm working in Xilinx Vivado (2024.1) with a fresh download of the core. I am using sim_clock_gate, sim_sffr, and sim_sffs. The only file that isn't taken from the repository is the test bench I'm using.
No inputs are X's. I've included every signal at the top level over the first 30 clock cycles as a full sanity check. I'm sure this is more than you asked for, but better safe than sorry!

[image]
[image]
[image]
[image]
[image]
[image]

Sorry about the incorrect cyclic signals, I was copying things over from my very messy personal notes and it seems something was lost in translation. For clarification:
- instr_req_o (cv32e40s_core.sv) comes from m_c_obi_instr_if.s_req.req.
- m_c_obi_instr_if.s_req.req (cv32e40s_instr_obi_interface.sv) is defined as either 1'b1 or trans_valid_i, depending on the current state.
- bus_trans_valid (fed into the obi interface as trans_valid_i in cv32e40s_if_stage.sv) comes from bus_trans_valid_o (cv32e40s_mpu.sv).
- bus_trans_valid_o (cv32e40s_mpu.sv) is assigned as "core_trans_valid_i && !mpu_block_bus".
- prefetch_trans_valid (fed into the MPU as core_trans_valid_i in cv32e40s_if_stage.sv) comes from trans_valid_o (cv32e40s_prefetch_unit.sv).
- trans_valid_o (cv32e40s_prefetch_unit.sv) comes from trans_valid_o (cv32e40s_prefetcher.sv).
- trans_valid_o (cv32e40s_prefetcher.sv) comes from fetch_valid_i.
- fetch_valid (fed into the prefetcher as fetch_valid_i in cv32e40s_prefetch_unit.sv) comes from fetch_valid_o (cv32e40s_alignment_buffer.sv).
- fetch_valid_o (cv32e40s_alignment_buffer.sv) comes from a combination of various signals (see lines 139 to 144), but we care about pop_q.
- pop_q (cv32e40s_alignment_buffer.sv) comes from "(instr_valid_o && instr_ready_i)".
- instr_ready_i (cv32e40s_alignment_buffer.sv) comes from prefetch_ready_i (cv32e40s_prefetch_unit.sv).
- prefetch_ready (fed into the prefetch unit as prefetch_ready_i in cv32e40s_if_stage.sv) comes from if_ready.
- if_ready (cv32e40s_if_stage.sv) comes from a handful of signals, but we care about predec_ready.
- predec_ready (cv32e40s_if_stage.sv) comes from "id_ready_i && !dummy_insert".
- id_ready (fed into the IF stage as id_ready_i in cv32e40s_core.sv) comes from id_ready_o (cv32e40s_id_stage.sv).
- id_ready_o (cv32e40s_id_stage.sv) comes from multiple signals, but we're concerned with ex_ready_i.
- ex_ready (fed into the ID stage as ex_ready_i in cv32e40s_core.sv) comes from ex_ready_o (cv32e40s_ex_stage.sv).
- ex_ready_o (cv32e40s_ex_stage.sv) comes from a ton of signals, but importantly alu_ready, csr_ready, and sys_ready.
- **alu_ready, csr_ready, and sys_ready (cv32e40s_ex_stage.sv) all come from wb_ready_i**.
- wb_ready (fed into the EX stage as wb_ready_i in cv32e40s_core.sv) comes from wb_ready_o (cv32e40s_wb_stage.sv).
- wb_ready_o (cv32e40s_wb_stage.sv) comes from lsu_ready_i, among other signals.
- lsu_ready_0 (fed into the WB stage as lsu_ready_i in cv32e40s_core.sv) comes from ready_0_o (cv32e40s_load_store_unit.sv).
- ready_0_o (cv32e40s_load_store_unit.sv) comes from "done_0 && !lsu_split_0_o".
- done_0 (cv32e40s_load_store_unit.sv) comes from ready_0_i, and other signals.
- lsu_ready_ex (fed into the LS stage as ready_0_i in cv32e40s_core.sv) comes from lsu_ready_o (cv32e40s_ex_stage).
- **lsu_ready_o (cv32e40s_ex_stage) comes from wb_ready_i, which is used to define alu_ready, csr_ready, and sys_ready**.

I hope this isn't too much information to be useful, I wanted to make sure the cyclic path is clear.

cnokes14, 2025-01-25:
In doing some further investigation, it appears my initial cyclic signals report was incorrect.
Instead, this error is generated by index-out-of-bounds errors occurring in cv32e40s_obi_integrity_fifo.sv and cv32e40s_lsu_response_filter.sv. Both fifo_q (cv32e40s_obi_integrity_fifo.sv) and outstanding_q (cv32e40s_lsu_response_filter.sv) are three elements long, but are indexed by two bit numbers with no protection.

I've implemented a hasty local fix (just stopping increments at 2'b10 and decrements at 2'b00), which seems to be preventing this bug from occurring. I can create a pull request some time in the next few days with a nicer fix.

Silabs-ArjanB, 2025-01-27:
Hi @cnokes14 (Apart from the out of bound indexing) do you still think there is a cyclic path? (I got an email with more detailed signals but I don't see that in this github issue)

cnokes14, 2025-01-27:
Hi Arjan,

No, I think the cyclic signals part was just a mistake on my end.
Following the path of X's through the core, they all end up originating from an out of bounds access. There's no sign of anything cyclic; my initial guess on cyclic signals probably just came from getting lost in similarly-named top level signals.